avrdudes/avrdude #2011: avrdude issues with pkobn_updi under FreeBSD 14.2 Release

mcuee, 2025-06-09:
This seems to work only once during the first insertion, tested with AVR64EA48 Curiosity Nano.

```
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
AVR part              : AVR64EA48
Programming modes     : SPM, UPDI
Programmer type       : JTAGICE3_UPDI
Description           : Curiosity nano (nEDBG) in UPDI mode
ICE HW version        : 0
ICE FW version        : 1.27 (rel. 129)
Serial number         : MCHP3428031800001161
Vtarget               : 3.31 V
PDI/UPDI clk          : 100 kHz

Partial Family_ID returned: "AVR "
Silicon revision: 2.1

AVR device initialized and ready to accept instructions
Device signature = 1E 96 1E (AVR64EA48)

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: unable to read from serial port (64)
Error: unable to open port usb for programmer pkobn_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: unable to read from serial port (64)
Error: unable to open port usb for programmer pkobn_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: unable to read from serial port (64)
Error: unable to open port usb for programmer pkobn_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: unable to read from serial port (64)
Error: unable to open port usb for programmer pkobn_updi

Avrdude done.  Thank you.
mcuee@freebsd14:~/build/avrdude_81_rc/tools $ avrdude -v -c pkobn_updi -p avr64ea48
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /home/mcuee/.avrduderc does not exist

Using port            : usb
Using programmer      : pkobn_updi
Error: unexpected reply to DAP_Info: 0x00 0x00
Error: short read, read only 64 out of 512 bytes
Error: unable to read from serial port (64)
Error: unable to open port usb for programmer pkobn_updi

Avrdude done.  Thank you.
```

mcuee, 2025-06-09:
@stefanrueger 

As this is not a regression and most likely not an avrdude issue but rather FreeBSD USB driver issue, this will not block avrdude 8.1 release.

stefanrueger, 2025-06-09:
Thanks, @mcuee. Interesting problem. @dl8dtl @MCUdude Is there anything AVRDUDE can do to before closing/exiting?

mcuee, 2025-06-11:
@dl8dtl 

Just wonder if you can reproduce similar issues under your FreeBSD machine with similar tools. Thanks.

mcuee, 2025-06-12:
Wiki updated.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-programmers

closed 2025-06-12 by mcuee