avrdudes/avrdude #2012: avrdude issue with urboot under OpenBSD 7.7

mcuee, 2025-06-09:
No issues with usbasp but somehow the test failed with urclock/urboot.

```
[mcuee@openbsdn4200 tools]$ sudo su
openbsdn4200# bash

[root@openbsdn4200 tools]# pwd
/home/mcuee/build/avr/avrdude_81_rc/tools

[root@openbsdn4200 tools]# ./test-avrdude -d0 -p "-c usbasp -p m328p" -v
Testing /usr/local/bin/avrdude 8.0-20250608 (de2d1ad5)
Prepare "-c usbasp -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���   0.351 s: fuse access: clear, set and read eesave fuse bit
���   0.317 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
���   0.531 s: chip erase
���   2.634 s: flash -U write/verify holes_rjmp_loops_32768B.hex
���   1.169 s: flash -T write/verify holes_rjmp_loops_32768B.hex
���   0.406 s: eeprom check whether programmer can flip 0s to 1s
���   3.011 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
���   5.813 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
���   1.392 s: chip erase and spot check flash is actually erased
���   0.397 s: spot check eeprom is erased, too

[root@openbsdn4200 tools]# avrdude -c usbasp -p m328p -U urboot:16mhz_autobaud_ce_ee -qq && echo OK
OK

[root@openbsdn4200 tools]# ./test-avrdude -d0 -v -p "-c urclock -P /dev/ttyU1 -p m328p"
Testing /usr/local/bin/avrdude 8.0-20250608 (de2d1ad5)
Prepare "-c urclock -P /dev/ttyU1 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���   2.721 s: chip erase
 8.0-20250608 (de2d1ad5)
���   5.820 s: flash -U write/verify holes_rjmp_loops_32768B.hex (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU1 -p m328p -Uflash:w:./test_files/holes_rjmp_loops_32768B.hex
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU1 for programmer urclock
Error: sp_list_ports() failed!
���   7.181 s: flash -U write/verify rjmp_loops_for_bootloaders_32768B.hex
���   2.897 s: flash -T write/verify holes_rjmp_loops_32768B.hex
���   1.727 s: eeprom check whether programmer can flip 0s to 1s
���   2.958 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
���   5.820 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU1 -p m328p -T "write eeprom ./test_files/holes_the_five_boxing_wizards_1024B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_1024B.hex:a"
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU1 for programmer urclock
Error: sp_list_ports() failed!
���   8.331 s: eeprom -T write/verify lorem_ipsum_1024B.srec
���   3.427 s: chip erase and spot check flash is actually erased

One or more AVRDUDE "-c urclock -P /dev/ttyU1 -p m328p" tests failed. Do you want to retry this particular test? (y/n): n
```

mcuee, 2025-06-09:
Test results for NetBSD 10.1 (dual boot on the same Intel N4200 CPU based mini PC.
* https://github.com/avrdudes/avrdude/discussions/1956#discussioncomment-13406490

```
[mcuee@netbsd10n4200 tools]$  ./test-avrdude -d0 -p "-c usbasp -p m328p" -v
Testing /usr/pkg/bin/avrdude 8.0-20250608 (de2d1ad5)
Prepare "-c usbasp -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���   0.302 s: fuse access: clear, set and read eesave fuse bit
���   0.268 s: fuse access: set eesave fusebit to delete EEPROM on chip erase
���   0.484 s: chip erase
���   2.554 s: flash -U write/verify holes_rjmp_loops_32768B.hex
���   1.109 s: flash -T write/verify holes_rjmp_loops_32768B.hex
���   0.354 s: eeprom check whether programmer can flip 0s to 1s
���   2.995 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
���   5.823 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
���   1.346 s: chip erase and spot check flash is actually erased
���   0.360 s: spot check eeprom is erased, too

[mcuee@netbsd10n4200 tools]$ avrdude -c usbasp -p m328p -U urboot:16mhz_autobaud_ce_ee -qq && echo OK
OK

[mcuee@netbsd10n4200 tools]$ ./test-avrdude -d0 -v -p "-c urclock -P /dev/ttyU0 -p m328p"
Testing /usr/pkg/bin/avrdude 8.0-20250608 (de2d1ad5)
Prepare "-c urclock -P /dev/ttyU0 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���   1.669 s: chip erase
 8.0-20250608 (de2d1ad5)
���   4.039 s: flash -U write/verify holes_rjmp_loops_32768B.hex
���   1.975 s: flash -T write/verify holes_rjmp_loops_32768B.hex
���   1.447 s: eeprom check whether programmer can flip 0s to 1s
���   2.770 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
���   4.509 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
���   3.132 s: chip erase and spot check flash is actually erased
```

mcuee, 2025-06-09:
Looks like an OpenBSD USB driver issue for Arduino Uno ATmega16U2 USB FW.

```
[root@openbsdn4200 tools]# avrdude -c urclock -P /dev/ttyU0 -p m328p -v
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /root/.avrduderc does not exist

Using port            : /dev/ttyU0
Using programmer      : urclock
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : Urclock
Description           : Urboot bootloaders using urprotocol
Protocol              : Urprotocol

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)

Avrdude done.  Thank you.

[root@openbsdn4200 tools]# avrdude -c urclock -P /dev/ttyU0 -p m328p -Uflash:w:./test_files/holes_rjmp_loops_32768B.hex:i -v
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /root/.avrduderc does not exist

Using port            : /dev/ttyU0
Using programmer      : urclock
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
Error: sp_list_ports() failed!

Avrdude done.  Thank you.

[root@openbsdn4200 tools]# avrdude -c urclock -P /dev/ttyU0 -p m328p -Uflash:w:./test_files/holes_rjmp_loops_32768B.hex:i -v
Avrdude version 8.0-20250608 (de2d1ad5)
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /usr/local/etc/avrdude.conf
User configuration file /root/.avrduderc does not exist

Using port            : /dev/ttyU0
Using programmer      : urclock
AVR part              : ATmega328P
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : Urclock
Description           : Urboot bootloaders using urprotocol
Protocol              : Urprotocol

AVR device initialized and ready to accept instructions
Device signature = 1E 95 0F (ATmega328P, ATA6614Q, LGT8F328P)
Auto-erasing chip as flash memory needs programming (-U flash:w:...)
specify the -D option to disable this feature
Erased chip
Reading 5459 bytes for flash from input file holes_rjmp_loops_32768B.hex
in 3 sections of [0x2000, 0x5fff]: 44 pages and 173 pad bytes
Preparing flash input for device bootloader
5760 bytes in 3 sections of [0x2000, 0x7e7f]: 45 pages and 0 pad bytes
Writing 5760 bytes to flash
Writing | ################################################## | 100% 0.96 s 
Reading | ################################################## | 100% 0.57 s 
5760 bytes of flash verified

Avrdude done.  Thank you.
```

mcuee, 2025-06-10:
> [@mcuee](https://github.com/mcuee), thanks! Is that restricted to `urboot` or a more general serial port or serial to USB issue USB comms issue with that board?

Let me check.

Two possible issues.
1) OpenBSD USB Stack issue
2) Arduino USB Serial FW issue

Arduino FW for their AVR related boards can be buggy and often not not updated for a long time.
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2

mcuee, 2025-06-11:
No issues under Linux: with Arduino Uno ATmega16U2 USB FW as USB-TTL serial converter.
```
mcuee@debian12j4125:~/build/avr/avrdude_8.1_rc/tools$ avrdude -c jtag2isp -p m328p -U urboot:16mhz_autobaud_ce_ee -qq && echo OK
OK

mcuee@debian12j4125:~/build/avr/avrdude_8.1_rc/tools$ ./test-avrdude -d0 -v -p "-c urclock -P /dev/ttyACM0 -p m328p"
Testing /usr/local/bin/avrdude 8.0-20250610 (9d7a893d)
Prepare "-c urclock -P /dev/ttyACM0 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.622 s: chip erase
✅   3.263 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   1.028 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   0.506 s: eeprom check whether programmer can flip 0s to 1s
✅   1.862 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   3.578 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   2.202 s: chip erase and spot check flash is actually erased
```

mcuee, 2025-06-11:
I will close this issue later if Optiboot has the same issue.

mcuee, 2025-06-11:
Optiboot is worse.

```
[mcuee@openbsdn4200 tools]$ sudo avrdude -c stk600 -p m328p -U ./optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex 
Reading 484 bytes for flash from input file optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex
Writing 484 bytes to flash
Writing | ################################################## | 100% 0.13 s 
Reading | ################################################## | 100% 0.08 s 
484 bytes of flash verified

Avrdude done.  Thank you.

[mcuee@openbsdn4200 tools]$ avrdude -c urclock -P /dev/ttyU0 -p m328p -x showall
0 0000-00-00 00.00  application 0 store 0 meta 1 boot 512 o8.0 --s-h-r-- vector 0 (RESET) ATmega328P

Avrdude done.  Thank you.

[mcuee@openbsdn4200 tools]$ ./test-avrdude -v -p "-c urclock -P /dev/ttyU0 -p m328p"
Testing /usr/local/bin/avrdude 8.0-20250610 (9d7a893d)
Prepare "-c urclock -P /dev/ttyU0 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���  12.729 s: chip erase
���  12.752 s: flash -U write/verify holes_rjmp_loops_32768B.hex
���   5.811 s: flash -T write/verify holes_rjmp_loops_32768B.hex (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -T "write flash ./test_files/holes_rjmp_loops_32768B.hex:a"
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
���   5.811 s: flash -T write/verify rjmp_loops_for_bootloaders_32768B.hex (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -T "write flash ./test_files/rjmp_loops_for_bootloaders_32768B.hex:a"
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
���   5.811 s: eeprom check whether programmer can flip 0s to 1s (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -Ueeprom:w:0x55:m -Ueeprom:w:0xaa:m
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
# ... the next test may therefore take longer
���   5.808 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -T "write eeprom ./test_files/holes_the_five_boxing_wizards_1024B.hex:a" -T flush -T "write eeprom ./test_files/holes_pack_my_box_1024B.hex:a"
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
���   5.811 s: eeprom -T write/verify lorem_ipsum_1024B.srec (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -T "write eeprom ./test_files/lorem_ipsum_1024B.srec:a"
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock
���   5.810 s: chip erase and spot check flash is actually erased (failed command below)
$ avrdude -qq -c urclock -P /dev/ttyU0 -p m328p -e -FAU flash:w:0xff:m -U flash:v:./test_files/holes_flash_0xff_32768B.hex
Warning: attempt 1 of 10: not in sync
Warning: attempt 2 of 10: not in sync
Warning: attempt 3 of 10: not in sync
Warning: attempt 4 of 10: not in sync
Warning: attempt 5 of 10: not in sync
Warning: attempt 6 of 10: not in sync
Warning: attempt 7 of 10: not in sync
Warning: attempt 8 of 10: not in sync
Warning: attempt 9 of 10: not in sync
Warning: attempt 10 of 10: not in sync
Warning: programmer is not responding; try -x strict and/or vary -x delay=100
Error: unable to open port /dev/ttyU0 for programmer urclock

One or more AVRDUDE "-c urclock -P /dev/ttyU0 -p m328p" tests failed. Do you want to retry this particular test? (y/n): n
```

stefanrueger, 2025-06-11:
@mcuee Have you tried other baud rates with autobaud than 115200? Technically, this baudrate is out-of-specs for 16 MHz Fcpu.

mcuee, 2025-06-12:
> [@mcuee](https://github.com/mcuee) Have you tried other baud rates with autobaud than 115200? Technically, this baudrate is out-of-specs for 16 MHz Fcpu.

I have tried 38400bps and 250000, both are working.

mcuee, 2025-06-12:
9600bps is also working.

```
[mcuee@openbsdn4200 tools]$ sudo avrdude -c stk600 -p m328p -U urboot:16mhz_9600baud_pr_ce_ee -qq && echo OK
OK

[mcuee@openbsdn4200 tools]$ ./test-avrdude -v -p "-c urclock -P /dev/ttyU0 -b 9600  -p m328p"
Testing /usr/local/bin/avrdude 8.0-20250610 (9d7a893d)
Prepare "-c urclock -P /dev/ttyU0 -b 9600  -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
���   2.401 s: chip erase
���  15.059 s: flash -U write/verify holes_rjmp_loops_32768B.hex
���   8.259 s: flash -T write/verify holes_rjmp_loops_32768B.hex
���   1.719 s: eeprom check whether programmer can flip 0s to 1s
���   4.089 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
���   7.329 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
���   8.959 s: chip erase and spot check flash is actually erased
```

closed 2025-06-12 by mcuee

mcuee, 2025-06-12:
Wiki updated
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#os-support-limitations-for-avrdude-70-and-above

mcuee, 2025-06-12:
I have updated the Wiki page again.
https://github.com/avrdudes/avrdude/wiki/Known-limitations-of-avrdude#known-issues-for-bootloaders

mcuee, 2025-06-13:
Just one more sanity check on NetBSD 10.1.
```
[mcuee@netbsd10n4200 tools]$ ./test-avrdude -d0 -v -p "-c urclock -P /dev/ttyU0 -p m328p"
Testing /usr/pkg/bin/avrdude 8.0-20250610 (9d7a893d)
Prepare "-c urclock -P /dev/ttyU0 -p m328p" and press 'enter' or 'space' to continue. Press any other key to skip
✅   1.654 s: chip erase
✅   4.030 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   1.960 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   1.448 s: eeprom check whether programmer can flip 0s to 1s
✅   2.771 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   4.440 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   3.120 s: chip erase and spot check flash is actually erased
```

mcuee, 2025-06-13:
Side note to myself -- needs to fix the utf-8 font under OpenBSD/NetBSD. 

The following inside .bash_profile helps but the display under xfce-terminal and Firefox are still not correct.
```
export LANG="en_US.UTF-8"
PS1='[\u@\h \W]\$ '
```

mcuee, 2025-06-13:
Display under OpenBSD.
```
[mcuee@openbsdn4200 tools]$ ./test-avrdude -d2 -v -p "-c urclock -P /dev/ttyU0 -p m328p -x strict"
Testing /usr/local/bin/avrdude 8.0-20250610 (9d7a893d)
Prepare "-c urclock -P /dev/ttyU0 -p m328p -x strict" and press 'enter' or 'space' to continue. Press any other key to skip
✅   2.658 s: chip erase
✅   4.309 s: flash -U write/verify holes_rjmp_loops_32768B.hex
✅   2.099 s: flash -T write/verify holes_rjmp_loops_32768B.hex
✅   1.679 s: eeprom check whether programmer can flip 0s to 1s
✅   2.907 s: eeprom -U write/verify holes_pack_my_box_1024B.hex
✅   4.929 s: eeprom -T write/verify holes_{the_five_boxing_wizards,pack_my_box}_1024B.hex
✅   3.379 s: chip erase and spot check flash is actually erased

```

nomakewan, 2025-06-15:
> Display under OpenBSD 7.7 -- Chromium browser is okay. Firefox is not okay. Cannot see the ✅ .

While I am on Windows, I experienced a similar issue on Firefox a while back when a browser installation accidentally corrupted the system font for emoji. I was able to resolve the corrupted system font, but in the interim I noticed that only Firefox was affected--Chromium browsers continued to display emoji with no issues whatsoever.

Looking into it, Firefox actually has a configuration setting that defines which specific font is to be used to display emoji. Perhaps you could leverage that setting to resolve the Firefox issue?

https://www.reddit.com/r/firefox/comments/jr1lt9/comment/gbr20s6