avrdudes/avrdude #2028: "MPLAB" programmers can't write to Xmega `boot` memory

MCUdude, 2025-06-20:
This issue was [discovered during testing before the 8.1 release](https://github.com/avrdudes/avrdude/discussions/1956#discussioncomment-13524066). The problem is that programmers that use @MX682X's implementation of "MPLAB programmers (PICkit5, PICkit Basic, PICkit4 in MPLAB mode, SNAP in MPLAB mode) can't write to Xmega's `boot` memory.

> avrdude exposes two ways to address a flash address of these devices - through "flash" and "application"/"boot".
MPLAB is only using "application" and "boot". As MPLAB knows where what goes it can write those areas sequentially.
And as MPLAB is erasing the device before an upload, the boot memory is always empty, so there is no problem that it does only "Write boot loader section page"

> But I won't tackle this issue now as I need to figure out the "unknown" instruction and this will probably take some time.

The issue is easy to replicate:

```
$ avrdude -cpickit4_mplab_pdi -patxmega256a3bu -t
avrdude> read boot 0 32
Reading | ################################################## | 100% 0.03 s 
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> write boot 0 "hello world"
Caching | ################################################## | 100% 0.00 s 
avrdude> flush
Synching cache to device ... 
Writing | -------------------------------------------------- | 0% 0.06 s 
Error: verification mismatch at flash page addr 0x40000
avrdude> read boot 0 32
Reading | ################################################## | 100% 0.00 s 
0000  68 65 6c 6c 6f 20 77 6f  72 6c 64 00 ff ff ff ff  |hello world.....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q
Synching cache to device ... 
Writing | -------------------------------------------------- | 0% 0.06 s 
Error: verification mismatch at flash page addr 0x40000
Synching cache to device ... 
Writing | -------------------------------------------------- | 0% 0.06 s 
Error: verification mismatch at flash page addr 0x40000

Avrdude done.  Thank you.

$ avrdude -cpickit4_mplab_pdi -patxmega256a3bu -t
avrdude> read boot 0 32
Reading | ################################################## | 100% 0.03 s 
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude> q

Avrdude done.  Thank you.
```

stefanrueger, 2025-06-20:
Am I right thinking this is not our bug? The programmer software/scripts don't cater for writing to the boot addresses of flash. The AVRDUDE project has an excellent history of making buggy programmers work, and it looks like @MX682X has an idea to do this here (great!) but we might want the label to reflect this. To be clear: I absolutely encourage ensuring AVRDUDE ironing out the programmer's bug, but if so we'd fix "not our bug".

MCUdude, 2025-06-20:
From @MX682X's response earlier, my understanding was that MPLAB supports writing to `boot`.

> avrdude exposes two ways to address a flash address of this devices - through "flash" and "application"/"boot".
MPLAB is only using "application" and "boot". As MPLAB knows where what goes it can write those areas sequentially.
And as MPLAB is erasing the device before an upload, the boot memory is always empty, so there is no problem that it does only "Write boot loader section page"

closed 2025-06-23 by stefanrueger