avrdudes/avrdude #2039: github avrdude_packing_release action failed

mcuee, 2025-06-24:
Somehow we did not have this issue previously for 8.0 release.

Last time a single dist.zip file will be used for all the builds. Now it seems that we need to change to name each build differently based on the architecture.

```
Run actions/upload-artifact@v4
/usr/bin/docker exec  bee507e0733f4a79dd7690dd2a4c4fe7898de9[1](https://github.com/avrdudes/avrdude/actions/runs/15824523096/job/44646408235#step:7:1)44c6d60d7a7d9f69e4e6f282c sh -c "cat /etc/*release | grep ^ID"
With the provided path, there will be 4 files uploaded
Artifact name is valid!
Root directory input is valid!
Error: Failed to CreateArtifact: Received non-retryable error: Failed request: (409) Conflict: an artifact with this name already exists on the workflow run
```

mcuee, 2025-06-24:
@umbynos 

Sorry to disturb you. Last time you helped on this github action. Just wondering if you can take a look to see what is the potential problem. Thanks.

mcuee, 2025-06-24:
It seems to me that somehow each run create the same `dist.zip` file and that may be the issue.

On the other hand, the other routine github action seems to work fine. It creates 6 `dist_xxxOS_xxxarch.zip` files.
https://github.com/avrdudes/avrdude/actions/runs/15824598803

[image]

umbynos, 2025-07-08:
Hey @mcuee, a little late but the problem should be that the `upload-artifact` uses an old version of the action. To upgrade to a newer version I recommend using this guide: https://github.com/actions/upload-artifact/blob/main/docs/MIGRATION.md

mcuee, 2026-04-17:
> Is this still an active issue [@mcuee](https://github.com/mcuee)?

Yes. It will be good that we to fix this before 8.2 release.

mcuee, 2026-04-18:
PR #2101 has been raised.

closed 2026-05-04 by mcuee