avrdudes/avrdude #2043: Add support to the Bus Pirate for larger flash chips

therealdreg, 2025-07-07:
I want to add support for larger flash chips to the modern Bus Pirate. I’ve opened this issue to coordinate with you on the best way to do it (#459):

https://github.com/avrdudes/avrdude/blob/main/src/buspirate.c#L989

btw, this is the modern bus pirate 

[image]

I’ve added support by porting all the code from Bus Pirate v3 and it’s already working! But now I want it to work with AVRDude for larger flash chips—what would be the best way to do that? Let me know what you’d like me to change in the firmware and I’ll update it.

Maybe the simplest solution is for me to send you a new version number here, for example: `CDC_SEND_STR(1, "\x02\x00\x01");` — and when that happens, you can send me new commands to read larger flash chips. Please let me know what you want me to modify in that file so everything is as simple as possible.

https://github.com/DangerousPrototypes/BusPirate5-firmware/blob/main/src/binmode/legacy4third.c#L657

[image]

@mcuee @MCUdude Do you want me to send you a modern Bus Pirate as a gift so you can test the new releases, like we did in issue  #1616 ?

mcuee, 2025-07-09:
With regard to the FW version handling, I think @stefanrueger should be able to advise.

therealdreg, 2025-07-09:
> With regard to the FW version handling, I think [@stefanrueger](https://github.com/stefanrueger) should be able to advise.

awesome! @stefanrueger tell me what is the simplest way

therealdreg, 2025-07-09:
> The older Bus Pirate implementation does not support memories with *_page sizes_ larger than 512. But that's not an issue because there doesn't exist any AVR that has this. But as proven in [#459](https://github.com/avrdudes/avrdude/issues/459), there is in fact an issue with larger chips.

Exactly, the idea is to fix this. As you can see, the SPI C code in the firmware is super easy to understand—I can adapt whatever you need.

Sending a PAGE SIZE `0xFFFF` First as a dummy-hack is just an idea to support both old and new versions of AVRDude at the same time (while allowing the legacy Bus Pirate v3 firmware to keep working as always…).

therealdreg, 2025-07-09:
@MCUdude @mcuee  This time the product is a gift from Ian @DangerousPrototypes  — he will ship it VAT prepaid, don’t worry, we always cover the VAT cost for free items. He’s covering all the expenses!

mbrugman67, 2025-07-12:
@MCUdude @mcuee @stefanrueger  @therealdreg - I'm jumping in here from the Bus Pirate forums where I saw this issue pop up.

I took at look at the Bus Pirate side of this - I had a suspicion it was due to the Bus Pirate code not handling the `extended high byte address (RAMPZ)` part of the addressing.

Taking a look at the code, I found this:  
```  
while (len > 0) {
    if (button_get(0)) {
        return;
    }

    hwspi_write_read(0x20); // AVR_FETCH_LOW_BYTE_COMMAND
    hwspi_write_read((addr >> 8) & 0xFF);
    hwspi_write_read(addr & 0xFF);
    uint8_t byte_flash = hwspi_write_read(0x00);
    if (binmode_debug) {
        printf("\r\n0x%02X", byte_flash);
    }
    tud_cdc_n_write_char(1, byte_flash); // Send the readed byte
    tud_cdc_n_write_flush(1);
    len--;
    if (len > 0) {
        hwspi_write_read(0x28); // AVR_FETCH_HIGH_BYTE_COMMAND
        hwspi_write_read((addr >> 8) & 0xFF);
        hwspi_write_read(addr & 0xFF);
        uint8_t byte_flash = hwspi_write_read(0x00);
        if (binmode_debug) {
            printf("\r\n0x%02X", byte_flash);
        }
        tud_cdc_n_write_char(1, byte_flash); // Send the readed byte
        tud_cdc_n_write_flush(1);
        len--;
    }
    addr++;
}
```  
So, yes; no extended high byte address is ever set.  I see that avrdude is sending address and len as a uint32_t, so it's just a matter of pulling the 3rd byte out and setting it:

```  
while (len > 0) {
    if (button_get(0)) {
        return;
    }
    hwspi_write_read(0x4d); // AVR_LOAD_ADDRESS_EXTENDED_HIGH_BYTE_COMMAND
    hwspi_write_read(0x00);
    hwspi_write_read((addr >> 16) & 0x03);  // just the two lowest bits
    hwspi_write_read(0x00);

    hwspi_write_read(0x20); // AVR_FETCH_LOW_BYTE_COMMAND
    hwspi_write_read((addr >> 8) & 0xFF);
.....
```  

Initial testing shows this resolves the issue - Dreg was able to read the full flash from an ATMega2560 part.  I also tested against an ATMega328p to be sure the `0x4d` command doesn't cause issues on parts with 64K or less flash.

What has your experience been with the `0x4d` command being used on smaller flash size parts?  Any issues?  From my AVR experience, I think this is all we need to do to get the Bus Pirate to correctly read flash beyond 64k words.

Just as a side note - I've been using avrdude for many years, typically with an AVR dragon or parts that have the Arduino bootloader.  I just want to say thanks for developing and maintaining such a nice tool! :)

stefanrueger, 2025-07-15:
ISP programmable AVR parts normally(!) provide the load extended address command 0x4d whether they have more than 64k flash or not (eg, both the m328p and the m1284p document the 0x4d command).

Some older ISP parts might not know the 0x4d command, though. I would carefully check the data sheets of those ISP parts that the buspirate wants to be able to program.

It might be safer for the programmer firmware to issue the 0x4d command only on parts that have flash in excess of 64k (these defo know that command). Also, it is only necessary to issue the 0x4d command when the extended address byte actually changes, ie, when reads or writes change 64k segments. AVRDUDE knows about holes in .hex files and can make jumps between segments, so the firmware cannot rely on sequential programming. Also, AVRDUDE by default verifies the flash contents, so after writing into a high segment will go back to reading from lower segments.

So yes, given the caveats above, changes in the firmware should be sufficient to make buspirate deal with larger ISP parts without changes to AVRDUDE.

therealdreg, 2025-07-15:
Thank you very much for your response and attention to detail @stefanrueger Could you take a quick look at this change and let me know your opinion, please? I don’t want to break anything or make things worse than they already were.

https://github.com/mbrugman67/BusPirate5-firmware/blob/1ec587488b60dc821d4d720d9ccbfcd27ab8ee2a/src/binmode/legacy4third.c#L698

mbrugman67, 2025-07-15:
I can add a conditional to only do the extended high byte write when flash size is greater than 64k words. Do we know the part/flash size in the BP after initial connect?  I can check when I'm back home next week.

therealdreg, 2025-07-15:
> I can add a conditional to only do the extended high byte write when flash size is greater than 64k words. Do we know the part/flash size in the BP after initial connect? I can check when I'm back home next week.

That’s why I’m asking @stefanrueger what he thinks about your changes… our code is pretty “dumb and I want the best possible solution. Depending on what they tell us, we’ll decide what to do.