avrdudes/avrdude #2046: program enable instruction not defined for part ATtiny20

jimktrains, 2025-07-25:
When trying to program an ATtiny20 (via ScratchMonkey on an Arduino Pro Mini), I am getting the following error:

> program enable instruction not defined for part ATtiny20

This happens in both avrdude v7.1 and v8.1. I'm unsure of how to proceed or what I can debug on my end. Would a different programmer be required? 

Thank you for any help. Below is the output from a run. (I'm using a binary in ~/Downloads because I fetched v8.1 from github to see if this was changed/fixed in the latest version.)

```
% ~/Downloads/avrdude_Linux_64bit/bin/avrdude -C ~/Downloads/avrdude_Linux_64bit/etc/avrdude.conf -P /dev/ttyUSB0 -c stk500v2 -p t20 -vv -U flash:w:build/prrplsattiny20.X.production.hex
Avrdude version 8.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/jim/Downloads/avrdude_Linux_64bit/etc/avrdude.conf
User configuration file /home/jim/.avrduderc does not exist

Using port            : /dev/ttyUSB0
Using programmer      : stk500v2
Input file build/prrplsattiny20.X.production.hex auto detected as Intel Hex
AVR part              : ATtiny20
Programming modes     : TPI

Memory       Size  Pg size
--------------------------
flash        2048       16
fuse            1       16
lockbits        1       16
io             64        1
sram          128        1
calibration     1       16
signature       3       16

Variants       Package  F max   T range        V range       
-------------------------------------------------------------
ATtiny20-CCU   UFBGA15  12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-CCUR  UFBGA15  12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-MMH   VQFN20   12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-MMHR  VQFN20   12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-SSU   SOIC14   12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-SSUR  SOIC14   12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-UUR   WLCSP12  12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-XU    TSSOP14  12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]
ATtiny20-XUR   TSSOP14  12 MHz  [-40 C, 85 C]  [1.8 V, 5.5 V]

Programmer type       : STK500V2
Description           : Atmel STK500 v2
Programmer model      : SCRATCHMONKEY
HW version            : 0
FW Version Controller : 2.00
Topcard               : Unknown
Vtarget               : 5.0 V
Varef                 : 5.0 V
Oscillator            : Off
SCK period            : 8.7 us
XTAL frequency        : 7.372800 MHz
Error stk500v2_program_enable() stk500v2.c 1030: program enable instruction not defined for part ATtiny20
Error main() main.c 1632: initialization failed  (rc = -1)
 - double check the connections and try again
 - use -B to set lower the bit clock frequency, e.g. -B 125kHz
 - use -F to override this check

Avrdude done.  Thank you.
```

mcuee, 2025-07-25:
ScratchMonkey claims to support TPI but it does not list ATtiny20. It would be good to confirm with another STK500v2 based programmer.
https://microtherion.github.io/ScratchMonkey/TPI.html

stefanrueger, 2025-08-11:
I had a quick look through `stk500v2.c` and can confim the code for this command-line wants to program ISP, not TPI. Looks like TPI programming is supported by the `stk500v2.c` source file for STK600, AVRISP MKII or JTAGICE MKII programmers but not for the STK500 programmer. Scratchmonkey has been added to `avrdude.conf` as STK500 compatible programmer, which explains the issue at hand.

I had another quick look through the scarce documentation at https://microtherion.github.io/ScratchMonkey/TPI.html and didn't find any evidence of tested TPI programming (though that may well work). In absence of better documentation it isn't clear how AVRDUDE should elicit TPI programming from ScratchMonkey.

jimktrains, 2025-08-11:
@stefanrueger That makes sense. Thank you for taking a peek at it.

closed 2025-08-11 by jimktrains