avrdudes/avrdude #2052: v8.1 support of Adafruit HV UPDIFriend clone

owenduffy, 2025-09-05:
I have a clone of the Adafruit HV UPDIFriend which uses a CH340E chip.

Testing on a t1614 target.

Monitoring with a scope, it seems that the only way I can get it to reliably send the HV pulse is the following:

avrdude -c serialupdi -x rtsdtr=low ...

It seems that the default (ie not specifying rtsdtr option) sends the HV pulse once or twice after plugging into USB, and then no pulses... except occasionally it does.

Possible solutions include:
- default to the rtsdtr=low behavior;
- create a programmer type serialupdihv which does the above.

Thanks
Owen

owenduffy, 2025-09-05:
I hasten to add that in my tests of the Adafruit HV UPDIFriend, I do not think there is a way to safely use it on a chip that would be damaged by the HV pulse. It seems to send the HV pulse uncommanded, eg when plugged into USB.

Owen

stefanrueger, 2025-09-07:
| default to the `rtsdtr=low` behavior

The avrdude project tries to be backward-compatible, so there needs to be a very good reason to break that. That could only been considered if *all* physical UPDI programmers that work without the `-x rtsdtr=low` *also* work with the `-x  rtsdtr=low` option. @MCUdude or @mcuee might comment on that.

Maybe it's a matter of documentation? @owenduffy Please feel free to suggest a helpful change of the current documentation.

I am not sure what's involved in creating a programmer variant but that would need documenting, too. As that's merely a shortcut for `-c serialupdi -x rtsdtr=low` it would enlarge the AVRDUDE's programmer space without adding functionality. Someone using a clone of Adafruit HV UPDIFriend with CH340E would _still_ need to figure out how to program but *everyone else* has one `-c` programmer more to consider, so balance I tend to think that's not helpful.

owenduffy, 2025-09-09:
serialupdi as defined by https://github.com/microchip-pic-avr-tools/pymcuprog/tree/main does not use RTS.

[image]

hv serial updi uses the RTS line to pulse 12V to the UPDI pin.

[image]

For this to work properly, AVRDUDE needs to assert RTS prior to each programming operation, and release RTS at the completion of the operation.

It can be made to work using programmer serialupdi and -x rtsdtr=low

Lots of people seem to trip up on this.

It would be better if AVRDUDE had a programmer type hvserialupdi that defaults to -x rtsdtr=low behavior.

Owen

owenduffy, 2025-09-10:
Here is an extract from the schematic for the Adafruit HV serialupdi.

[image]

RTS from the CH340E USB serial chip connects to the switch chip at the right to pulse 12V onto the UPDI pin when /RTS goes low.

This circuitry does not exist in the non-HV version of the serialupdi, RTS is not used.

BUT asserting RTS is ESSENTIAL to the HV serial updi operation to reconfigure UPDI mode on a blocked chip.

Asserting RTS should be automatic (default), and if needs be, a modified programmer type created.

Owen